ponzano/MeshLink-hardware #1: ADC Divider

ctmalan, 2025-02-20:
2 : 3 ratio of the divider is better as this is the default in the settings.  I use 2M and 3M with 100nF smoothing cap.  (10 uF might be overkill?)

c-:

ponzano, 2025-02-20:
> 2 : 3 ratio of the divider is better as this is the default in the settings.  I use 2M and 3M with 100nF smoothing cap.  (10 uF might be overkill?)
> 
> c-:

Hi! Thank you for your interest 😊
I chose to use 4M and 10M to increase resolution on the adc pin, as it gives 3V when battery is at 4.2V instead of 2.52V. Maybe it isn't necessary as the adc is 12bit but I thought it's nice to optimize the range anyway. The value is anyway taken into account in the variant.h file and has been manually calibrated with a precise multimeter, so the final user doesn't need to worry about that. 
In any case, the voltage divider is redundant now because there's an ina219 to read battery voltage too. 
The capacitor could be smaller but cost wise it doesn't make any difference, it's selected from the basic components library of JLCPCB to keep cost down (extended library parts cost an extra fee of 2.6€ iirc for each component) 
Are there any downsides to my method I might not have thought about? As I always say, I'm open to constructive feedback and very happy to have received this issue 😌🩵
Thank you

ponzano, 2025-02-20:
Uh, for additional context: the calibration has been performed on an actual prototype and functionality has been tested for several months as of now, we haven't encountered any issues (Yet XD)

ctmalan, 2025-02-20:
I have preference for the Meshtastic default ratio.  Less things to change.

closed 2025-02-25 by ponzano