ponzano/MeshLink-hardware #2: Option for 4 solar panels

ctmalan, 2025-02-20:
Tree hanging nodes need four panels.

Refer to:  https://preview.redd.it/in-honour-of-the-other-tree-hanging-node-v0-r1isbzq37dgd1.jpg?width=640&crop=smart&auto=webp&s=8355d1b8e7cbfc586ecf6623cfc73ab6adc2d593

ponzano, 2025-02-20:
Hmmmm interesting concept, I'll try to think if and how it could be implemented. 
But is it actually necessary? 
We have nodes based on nrf52 deployed in the area with a single small 5V panel that have more than enough power overhead. 
Instead of using 4 separate panels maybe it could be an option to use a slightly bigger 12v or 18v one? They would have much higher maximum power, which means that even with bad insulation they'd provide enough power to charge the battery. Consider that the board consumes on average around 7-11mA.
In any case, I'll see what I can do because it sounds cool 😁

ctmalan, 2025-02-20:
Or with 4 x Shottky's on the input side.  Components can be fitted as needed.  Might also double as the blocking diode discussed elsewhere?